jezs00/pycasso #6: Feature Request: Additional Pijuice scripts

jezs00, 2022-11-20:
As pijuice has 3 buttons, could have additional functions run from each.

* Function to delete last image and fetch a new one to avoid garbage being saved and used again in historic. Would require logging of path of last saved image.

recrudesce, 2023-01-19:
Thoughts:

- you'd only need to log where the installation went to.  because the images are always in `images/generated`
- have a shell script that just goes to the images directory and deletes files that are X days old.
- either cron that shell script, or associate it with the button on the PiJuice

jezs00, 2023-01-19:
Good thoughts. Something I need to test on the PiJuice is whether it can
run a script based on the button pressed, even if that button is the one
starting the system. I figure it should be possible.

recrudesce, 2023-01-20:
So I don't use a PiJuice, I use another device called a PiSugar 3 Plus because I couldn't get a PiJuice in the UK for love nor money (fun fact, I wrote a handler for the PiSugar, so when I've confirmed it all works, I'll give that to you too), which has a button on it that you can make run scripts - it has a single tap, a double tap, and a long tap option.  It wouldn't be that hard at all to make a script that just cleans out the images/generated folder, and then assign it to the button on the PiSugar.  Looks like it's possible on the PiJuice too, because their product page says "trigger events and scripts with the thre programmable buttons".

https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series is what I have.

jezs00, 2023-01-20:
Looks good, keen to support as much hardware as possible and keep the app modular. Let me know how you go.