jezs00/pycasso #23: Implement Stable Horde

jezs00, 2023-01-17:
Suggestion to implement https://stablehorde.net/ . Need to investigate.

recrudesce, 2023-02-10:
I've been looking into this.  You really only need to deal with 3 API endpoints:
 - Make me an image
 - Is my image done yet ?
 - Give me my image

Images can apparently be queued for 10 minutes unfulfilled until they are deleted from the queue.  So you'd need to implement a process that 
 1. asks for the image (easy), `/v2/generate/async`
 2. regularly check to see if the image is done yet (can take up to 10 minutes), `/v2/generate/check/{id}`
 3. get the image if it's done, or do something if it's not `/v2/generate/status/{id}`

I'm thinking you could just take the sdwebui library and modify it for StableHorde, but I just have worry about locking up a thread to just check for the image completion and just burning battery life while waiting.

jezs00, 2023-02-11:
It's a bit of work but yes that'd be definitely possible. A couple of ideas about saving battery:
1 - You could program pijuice or your battery handler to request the image, save the ID, shut down, start up again in x minutes, check if the image is there, get the image, shut down and start up again back at the normal time.
2 - Depending on how long stable horde keeps the image for, you could also pre-cache the prompt and request it the day before. So every day you are setting the image to what was requested yesterday. Would need some persistence to save the progress or last id from stable horde.

Both options should be possible but would be a bit of a project to get working and a bit more prone to error/failure than the other methods.

Efreak, 2024-04-06:
Horde has an SDK [now](https://github.com/Haidra-Org/horde-sdk) if you don't want to implement this yourself.

> 2 - Depending on how long stable horde keeps the image for.....

Horde only keeps results for ~20 minutes after they're done, you'll need to keep them yourself after that. Webhooks were recently added, so that can be used if there's a portable webserver built-in here (I haven't looked at this project much, I'm just looking for more horde stuff on GitHub)