jezs00/pycasso #25: Implement Stable Diffusion local web

jezs00, 2023-01-17:
Create provider wrapper to look for this webui running locally (or on the internet, just provide a host): https://github.com/AUTOMATIC1111/stable-diffusion-webui

jezs00, 2023-01-30:
This exists, could help.
https://github.com/mix1009/sdwebuiapi

recrudesce, 2023-02-02:
See #38

jezs00, 2023-02-03:
Awesome, it works great! Good job, and code is really clean, it just works straight out of the box. I'll throw it in a try/except to support future work in retrying on provider failure, and you are right about my love of pulling things from config, I'll set up a Host/Port in config to keep people from having to mess with code if they don't want to.

jezs00, 2023-02-03:
There was just a minor bug on fetch_height/fetch_width not being used, which means items that are not a valid multiple might be smaller than the screen. Fortunately the Automatic1111 multiple is only 8 so that's already coded, I will just update the constant and implement fetch height/width.

commit 1422da55 (2023-02-03): Bumped version number. Closes #25
closed 2023-02-03 by jezs00