jezs00/pycasso #26: Add hierarchy of providers as backup

jezs00, 2023-01-17:
Add a hierarchy of providers/modes to default to if other modes fail.

Will require more robust try/except within providers.

If any mode fails, we can display a shape on the new image. I'm thinking a circle.

recrudesce, 2023-01-20:
Now that you're adding a battery icon, maybe have a little status bar at the top that houses the battery icon, and then a web (if using online generation) or a little disk icon (if using local files).  That way at a glance you can see the battery state, and also where the currently displayed image came from ?

jezs00, 2023-01-20:
Sounds good. Considering we have a lot of possible icons we want to use:

- Low battery warning (Currently square. Can probably replace with a nice icon from flaticon, for people who don't choose to have battery level always showing)
- PiJuice failed to load (Currently X)
- Battery level
- Disk/Web/Test
- Notification of failure to load default mode

We would, as you say, like to collect these in an invisible task bar. This is a little complicated as we can't just define a location for each image, we would have to collect the icons we want to display, then display them in order, EG right to left.

I think the possibility for extension to this makes sense to create a new icon_handler.py and offload all icon management to this. This way the icon management object can be created at the start of the program, and whenever you want to add an icon, we can use an add_icon(icon) function to add it to the list. Internally it will hold a list of icons. Finally when we want to actually draw the image, we can pass the image object to a draw_icons(image) function and it handles the logic of drawing the icons.

constants.py to hold icon types and file paths.

recrudesce, 2023-01-20:
you have a little space at the top/bottom of the image, because it seems to crop the images in such a way that leaves a black bar at the top/bottom (at least all the ones that i've generated have).  so maybe just make the icons 16x16 and put them in a little bar at the top like Android/iOS does on phones ?

Agree with all other sentiments as well - palm it off to another handler, definitely.

jezs00, 2023-01-20:
I'd probably just put the images in with configurable opacity so users can decide how they'd like to see it. If users are getting an image that's not cropped, it's going to be different for different external image sizes, different screens and different providers. Something with configurable opacity or color will probably work for everyone, I could probably also config out the icon size so you can choose whatever suits your build.

>it seems to crop the images in such a way that leaves a black bar at the top/bottom 
What provider are you using, or are you using external images? Ideally I was hoping to always have an option for all users to completely fill their frame, whether through cropping or infilling.

recrudesce, 2023-01-21:
> What provider are you using, or are you using external images? Ideally I was hoping to always have an option for all users to completely fill their frame, whether through cropping or infilling.

Stability - I always get a small black bar at the top and bottom for any generated image.

[image]

jezs00, 2023-01-21:
That's annoying, Stable Diffusion should just fit. I've logged #32 to investigate, will look into it.

commit 8cb715e0 (2023-02-05): Added fallback function. Closes #26
closed 2023-02-05 by jezs00