jezs00/pycasso #27: ChatGPT to generate prompts

recrudesce, 2023-01-18:
I just ordered the kit to make one of these, cos it's amazingly cool.  I've been messing around with AI stuff, image generation AND text generation, and thought it'd be interesting if pycasso could potentially use an external API to generate prompts.

In my research I've found that while there are external prompt generators, most of them do not provide an API.  Then I came across this: https://pub.towardsai.net/design-your-ai-art-generator-prompt-using-chatgpt-7a3dfddf6f76

So I'm wondering how easy it might be to implement something like this - I'll give it a go when my hardware arrives, but you might have better luck considering how more advanced you are with Python than I.

jezs00, 2023-01-19:
This would definitely be possible, and to be honest probably not that hard. We already use openai's python api wrapper, and chatgpt will likely be rolled into it, or implemented in another wrapper. I'll leave this for now, and can pick it up when the API is released.

jezs00, 2025-05-11:
This is actually done now in release 1.3.0. See readme on how to use.

closed 2025-05-11 by jezs00