jezs00/pycasso #29: Review_Screen Only in B&W, but waveshare epd program in colour

recrudesce, 2023-01-19:
Hey, so I have followed instructions, installed everything, compiled the little EPD test program as per the Waveshare documents, and it shows the pictures in colour.

But when I run the review_screen.py code, or test with omni-epd-test, the picture only shows up in B&W.

Any thoughts ?  I have the correct display type set in the config, and it's reading it correctly out of the config, but the picture never shows up in colour.

recrudesce, 2023-01-19:
OK, quick update.  I reflashed the SD card, started again, and I'm still only getting b&w through omni_epd test tool and Pycasso.
It seems to completely ignore the mode, and just sticks to B&W.  Is there a way to force the mode to colour ?

jezs00, 2023-01-19:
Reopening this issue as I think I know what happened here.

You _shouldn't_ need to run the waveshare setup. omni-epd uses config to specify color mode, but that only occurs if it's loaded its config.ini, which we remove and replaced with .config in this project. I think it's possible that the review_screen.py was not loading the config file properly and that resulted in default display type being used, and default color which is BW.

In either case, I think for an example script like review_screen it's better just to ask for input in command line to determine whether the user wants bw or color. I will close once I've implemented and tested that.

closed 2023-01-20 by jezs00

recrudesce, 2023-01-20:
Confirmed this works perfectly - test images show up in colour :)

commit b5ba90b6 (2023-01-20): Script now sends config dict to epd when creating. Should help with issues on first run. Exits if config file not found. Fixes #29