jezs00/pycasso #31: Multiple Post Scripts

erbmur, 2023-01-20:
Would it be possible to change the postscript from string to strings?  That way we can have other options then just "digital art, trending on artstation"?
Or is there already a way to do this?

jezs00, 2023-01-20:
Do you mean have it randomly select from a group of possible post-scripts? Or you just want to change "digital art, trending on artstation" to something else?

erbmur, 2023-01-20:
Random post-scripts

jezs00, 2023-01-20:
Yeah, definitely. The easy way to implement would be to just parse the config string through the predefined custom markup function so that you could have something like:
`(digital art|painting), (trending on artstation|8K Ultra HD)`
which would give you 4 random possibilities of postscript:
_digital art, trending on artstation
painting, trending on artstation
digital art, 8K Ultra HD
painting, 8K Ultra HD_

The less easy way would be to create a new .txt file with several lines. It would still support markup like the above.
As a user, what do you like better?

erbmur, 2023-01-20:
First option would be fine, saves you the trouble of having to go through another txt file generation.

jezs00, 2023-01-21:
Done for prescript, connector and postscript.

closed 2023-01-21 by jezs00