jezs00/pycasso #32: Stable Diffusion Image Size

jezs00, 2023-01-21:
Check Stable diffusion works without resize and crops instead. Try to recreate https://github.com/jezs00/pycasso/issues/26#issuecomment-1399090090 .

recrudesce, 2023-01-21:
Could you not just ask SD to generate the image based on the known epd screen size ?

jezs00, 2023-01-21:
Yep that's what it's supposed to do. It accepts only multiples of 64 so
normal process is request an image slightly too big and then crop it. My
assumption is it's resizing somewhere instead of cropping.

jezs00, 2023-01-21:
Issue is based on resize_image() in provider.py. This will resize the image to capture as much as possible of it on the screen, which works depending on the display, however if the factor of resize required for the short edge of the screen is smaller than the factor of resize required on the long edge, it will display black bars. Writing replacement function fit_image() to thumbnail on a factor, instead of just on longest edge.

jezs00, 2023-01-21:
Should be fixed. Just testing on pi, then I'll bundle it into a release.

jezs00, 2023-01-21:
Appears to work. Closing for now and releasing new version.

closed 2023-01-21 by jezs00