jezs00/pycasso #33: Allow/Encourage Dithering

jezs00, 2023-01-21:
Should be able to get some better looking images on EPDs using dithering. https://github.com/robweber/omni-epd/wiki/Image-Dithering-Options

Considerations: Operations will probably be computationally heavy. Need to balance with battery life.

Realistically the option is already implemented as omni-epd already eats the config file, but if it makes the image look better it is worth putting into the readme and default config. So:

- Test out a few different dithering options with color EPD
  - Gauge processing time, whether it looks good
- Add best looking dither algorithm to .config-example

recrudesce, 2023-01-25:
This'll help dramatically.  It makes a lot of difference on the Inky Impression I have here:

https://github.com/pimoroni/inky/blob/master/examples/7color/advanced/dither.py

jezs00, 2023-01-25:
This looks great. I'll try and default one that is quick but looks good.

recrudesce, 2023-01-25:
So I added the dithering option to the config and adjusted the config_wrapper to pull the setting into the config dict, but once again all images are turning into black and white images.  It's weird, it's like it has a problem with the config and is reverting back to default settings.

I probably did something wrong :P

recrudesce, 2023-01-25:
OK, fixed it really easily.

in .config, add the following line under the `[EPD]` section (this is for 7 colour displays)
`palette_filter=[[0, 0, 0], [255, 255, 255], [0, 255, 0], [0, 0, 255], [255, 0, 0], [255, 255, 0], [255, 128, 0]]`

then under the `[Display]` section, add `dither=floydsteinberg`

It should just work.

jezs00, 2023-01-25:
Looks really good. I'll look at setting up some of these defaults today and just make sure it's not taking too long.

>  adjusted the config_wrapper to pull the setting into the config dict,

FYI you shouldn't need to modify config_wrapper for any omni-epd options as that class just consumes the entire config and reads the dict itself.

jezs00, 2023-01-25:
Wow I just tested it out and it does make things look much better, and according to the log, the floydsteinberg dithering is fast. I will put the dithering options in config-example but leave them commented out.

jezs00, 2023-01-25:
Closing issue. Can't believe I've gone this long without dithering!

closed 2023-01-25 by jezs00