jezs00/pycasso #37: Warning about the cost to run

recrudesce, 2023-01-26:
It might be worth mentioning somewhere that ultimately this won't be free to run.

Dreamstudio gives:
> All new users will get 100 free DreamStudio credits, enough to create 500 standard images. This is a one-time bonus when you first create your account, and will not replenish.

So once you've done 500 days of images (1.5 years if you only generate once a day at default settings (0.2 credits per generation), but you'll burn through that a lot quicker if you generate multiple times a day), it'll start costing you money to the tune of $10 per 1000 credits.

Ref: https://beta.dreamstudio.ai/faq

DALL-E gives:
> You get 50 free credits during your first month, and 15 free credits will refill every month after that.

Which is essentially 50 images in the first month, then 15 each month after that. Unused credits don't roll over, so you have to buy them at $15 per 115 credits.

Ref: https://help.openai.com/en/articles/6399305-how-dall-e-credits-work

Midjourney is charged based on GPU minutes, you get 25 minutes free as a one off, then you have to pay after that if you use them.  They never refill - once the free 25 minutes are gone, they're gone.

Ref: https://neuroflash.com/blog/midjourney-subscription/

jezs00, 2023-01-26:
I go into costs I've experienced in the article a little, but I think you're right and people should be reminded that each run costs money in some way. If the APIs support it we could use it to provide feedback on how much a certain run cost, and how much funds they have remaining in their account. Could even possibly include that in the status bar.

recrudesce, 2023-01-26:
The Stability API definitely doesn't allow you to query remaining credits, you'd just be guessing how much each generation cost based on the size of the image and the steps.

DALLE will only return the cost of a TEXT request in the response, not when doing an IMAGE request, bizarrely.

I don't think you can currently interact with Midjourney by any means other than their Discord bot, and they definitely do not have a public API available.  But it seems that they're planning on releasing one in the future.

I don't think we need to put something on screen that says "this cost you money", but rather just a disclaimer in the readme that says "yeah, ultimately this'll cost money, so if it stops working after a month (DALLE) or a year (Stability) the you probably wanna go buy some credits."

jezs00, 2023-01-26:
Yeah it's probably more of a 'nice to have'. If it becomes possible it would still be interesting to throw it in the logs. I really did expect midjourney to have an API by now, that's not implemented until they can get their act together. I'll update the readme.

commit 8cf25119 (2023-01-26): Update README.md

Added a note regarding paid providers closes #37
commit 24922853 (2023-01-27): Update README.md

Added a note regarding paid providers closes #37
closed 2023-01-26 by jezs00