jezs00/pycasso #45: review.py wrong config file location

JoghurtConnaisseur, 2023-02-07:
review.py has the wrong file location for the .config file. It's looking for it in .../.config/.config and not in .../pycasso/.config

`pycasso@raspberrypi:~/pycasso/examples $ python3 review_screen.py
WARNING:root:Unable to find file at '/home/pycasso/pycasso/.config/.config' or at backup path '/home/pycasso/pycasso/.config/examples/.config-example'
INFO:root:Loading config
Traceback (most recent call last):
  File "/home/pycasso/pycasso/examples/review_screen.py", line 33, in <module>
    display_type = config.get("EPD", "type")
  File "/usr/lib/python3.9/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.9/configparser.py", line 1149, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'EPD'`

jezs00, 2023-02-08:
Fixing up script. We might end up removing the review_screen.py as test mode in pycasso is doing its job better.

jezs00, 2023-02-08:
Fixed in cab144c

closed 2023-02-08 by jezs00