jezs00/pycasso #46: Crossed out battery symbol

JoghurtConnaisseur, 2023-02-08:
I’m getting this symbol on my display. What does it mean? It’s a crossed out battery symbol. My setup is using pijuice and auto power off after completion.

[image]

jezs00, 2023-02-08:
It's supposed to mean that pijuice class isn't loading properly and applying battery indicator, however it's probably an error seeing as I have the same. Will look into it. You can disable icons in .config if you like.

jezs00, 2023-02-08:
Issue was with range assignment, using - instead of ,

Appears to be working now, will create new release with this fix and the previous in it.

commit feb2d403 (2023-02-08): Fixed ranges in add_battery_icon fixes #46
closed 2023-02-08 by jezs00