jezs00/pycasso #52: Rotating screen to portrait

djxie2019, 2023-03-17:
With the rotate config setting to "90" I thought I could position the EPD in portrait orientation however crops the image instead. Then thought the in-fill request could fill those but doesn't seem to work. Is there another setting that can fix this to support portrait orientation?

jezs00, 2023-03-17:
Rotate is just a value used by omni-epd for the screen so pycasso doesn't
really handle it.

Solution probably could be as simple as:
* refactor use of Epd.width and Epd.height as individual variables
* add function to check rotate value in config and swap width and height
appropriately

Also add rotate unit test if implemented

jezs00, 2023-03-24:
I think this issue is innate to within the omni_epd functionality so for now to work with it, we have a new config variable called image_rotate in [Generation]. This will rotate the dimensions prior to requesting from providers.

closed 2023-03-24 by jezs00

djxie2019, 2023-03-24:
Have updated code today and set both the EPD and new Generation rotate settings to 90 degrees however still getting black bars top and bottom in portrait orientation - anything else I need to be setting?

jezs00, 2023-03-24:
Don't set Epd rotate to 90, set it back to 0

djxie2019, 2023-03-24:
In that case the image renders as per prior state in landscape orientation. I used option "2 Upgrade pycasso minimally" when updating - do I need to do a full install again?

jezs00, 2023-03-25:
Minimal update should work, but you can try full install if you want.
Make sure -

```
[Display]
rotate = 0

[Generation]
image_rotate = 90
```

djxie2019, 2023-03-25:
Completed a fresh install, still no luck. In the default config your "image_rotate" has a space before and after the "=" which none of the other parameters have but have tried both ways and still orientating in landscape.

jezs00, 2023-03-26:
Okay this should be fixed now if you update minimally, tested on pi4. Issue was:

- not rotating image back once first rotated
- version number not incremented

djxie2019, 2023-03-27:
Perfect, working now.

commit b7968e34 (2023-03-24): Updated config with new image_rotate variable. Refactored epd.width and epd.height. Added checking for rotation and swapping height/width. Added set_rotate() to handle. 2 new unit tests. Fixes #52

Updated readme to specify what image_rotate does. #52