jezs00/pycasso #54: "Multiple" image on resultant .png file

simonjowett, 2023-04-30:
Hi I'm getting 2 images on 1 picture.

[image]

The log for the promp is as follows:
```2023-04-30 16:19:07 INFO     Config loaded
2023-04-30 16:19:07 INFO     Starting program without using PiJuice
2023-04-30 16:19:07 INFO     pycasso has begun
2023-04-30 16:19:07 INFO     Requesting 'Portrait of an middle aged South American man, with short hair, in the style pixel art, trending on artstation'
2023-04-30 16:19:07 INFO     Loading Stability API
2023-04-30 16:19:07 INFO     Using grpc.stability.ai:443 as stability host
2023-04-30 16:19:07 INFO     Getting Image
2023-04-30 16:19:14 DEBUG    STREAM b'IHDR' 16 13
2023-04-30 16:19:14 DEBUG    STREAM b'eXIf' 41 612
2023-04-30 16:19:14 DEBUG    STREAM b'IDAT' 665 65536
2023-04-30 16:19:14 INFO     Saving image as /home/pi/pycasso/images/generated/pycasso - Portrait of an middle aged South American man, with short hair, in the style pixel art, trending on artstation.png
2023-04-30 16:19:15 DEBUG    STREAM b'IHDR' 16 13
2023-04-30 16:19:15 DEBUG    STREAM b'bKGD' 41 6
2023-04-30 16:19:15 DEBUG    b'bKGD' 41 6 (unknown)
2023-04-30 16:19:15 DEBUG    STREAM b'IDAT' 59 198
2023-04-30 16:19:15 DEBUG    STREAM b'IHDR' 16 13
2023-04-30 16:19:15 DEBUG    STREAM b'bKGD' 41 6
2023-04-30 16:19:15 DEBUG    b'bKGD' 41 6 (unknown)
2023-04-30 16:19:15 DEBUG    STREAM b'IDAT' 59 378
2023-04-30 16:19:15 INFO     Prepare epaper
2023-04-30 16:19:15 INFO     preparing omni_epd.mock
2023-04-30 16:19:15 DEBUG    Rotating image 0.0
2023-04-30 16:19:15 DEBUG    Applying contrast: 1.0
2023-04-30 16:19:15 DEBUG    Applying brightness: 1.0
2023-04-30 16:19:15 DEBUG    Applying sharpness: 1.0
2023-04-30 16:19:15 INFO     omni_epd.mock writing image to /home/pi/pycasso/mock_output.png
2023-04-30 16:19:16 INFO     Send epaper to sleep
2023-04-30 16:19:16 INFO     closing omni_epd.mock
```
If I use the prompt directly in the ai website it works fine.  I think it may be a scaling issue as I set the resolution to 800 x 480 as I propose to use a Waveshare 7.3" 800 x 480 color EPD .
```
[Debug]
#######################
# Debug Configuration #
#######################
# The following settings are only relevant for development. Only use them if you know what you're doing.

# Width and height to use with mock EPD, to test different display sizes while developing without an EPD.
test_epd_width = 800
test_epd_height = 480
```
If I set the resolution back to 600 x 488 the resulting png seems to be ok, ie not 2 heads!  Is there anything that can be modified to resolve this (if indeed it is a problem).  Maybe I can call it a new artform!

jezs00, 2023-04-30:
It'll have to be something happening in the AI provider itself, pycasso is just requesting the image for the dimensions you've provided. Some of the AI art providers can get confused with this kind of thing when your dimensions get further away from a square.

You can probably play around with your prompts to achieve better results, or try a different provider. DALLE with infilling is what I've found to be best for outpainting backgrounds for a more well spaced image at present and it's what I'm using. Otherwise you can just enjoy the weird things the providers do.

jezs00, 2023-09-03:
Closing due to inactivity, please let us know if this is still an issue that can be fixed in this project.

closed 2023-09-03 by jezs00