jezs00/pycasso #55: A few minor things and a thank you

hplato, 2023-07-06:
This was a great project, thank you for putting this together. I did this as a group project to show a fun outcome with some advance technology. 2 things that I came across when going through the steps.

1. For whatever reason, the bash install script didn't create the /home/pi/pycasso structure. So I had to do a git clone, and then run the bash script locally. It also didn't create the images/generated images/external or prompts/* for me. I had to create these afterwards. No big deal.

2. I find with the pijuice that it sometimes turns off the RTC wakeup. I think it's due to this: https://github.com/PiSupply/PiJuice/tree/master/Software in the repeated wakeup. I'm going to add that script to cron so that at powerup it will always reset the wakeup.

It might also be useful to show the pins you used on the waveshare. I purchased the same e-ink display and had to map the waveshare instructions to the colored wires, and then the colored wires to the PI pins. No big deal, but a reference might be helpful to compare against.

Again, thank you for packaging this together into a very cool project.

jezs00, 2023-07-09:
Thanks mate glad it is useful to you.

1. The bash script is hard to debug so this might be hard to fix. **If others have the same issue please post here so I can collect more information on it.** I have added this to the title so others may throw their feedback on this.
2. The python script itself should have code that re-enables the wakeup automatically, however if for whatever reason it doesn't make it to the end of the pijuice script it might not run. I'll keep an eye on it.

I did the same as you when I set up the EPD. I understand that it would be useful to provide the pins. Due to the nature of the program and instructions, we should be supporting any EPD, with this in mind I'd prefer to refer to vendor instructions (even if they aren't very good). In my opinion finding out what the pins do is probably a nice thing to learn about and really know what you're looking at before you go plugging things into it.

Have fun!

hplato, 2023-08-22:
One other question. So I mounted the PIs in frames, and got a short USB extender cable to run outside the frame so I can charge it up. The battery has run out, so I plugged it in. However, the PIJuice didn't wake up this morning to grab the new image. I don't want to take the frame apart each time the battery needs to be plugged in (to hit the on button on the PI Juice). Any ideas on how to configure the PIJuice so that it doesn't loose the wake up time?

I think the other PI that I have, if I plug in the microUSB on the PIJuice, it powers up the PI. Then I had to SSH in to power it down. I need to confirm that behavior tonight.

jezs00, 2023-08-23:
For mine, I find that default behavior of the pijuice is to automatically turn on once power is plugged in. As such, I just plug it in, wait for it to probably be charged and then unplug it and ssh in and restart the pycasso service which should both end up with a shutdown device and wakeup alarm reset. If your device doesn't power on after being plugged in, I am not sure what might cause that, try playing around manually with the pijuice cli config.

hplato, 2023-08-23:
I'm plugging mine in through the microusb on the PIJuice rather than the USB-C on the PI itself. I made a change to PIJuice, and set Wakeup on charge to 50%. Hopefully this powers on the pi at 50%.

One thing I've noticed is that a lot of the generated images have the 'interesting' part of the image at the top. So my banksy farmer that it just created, the head is cut off as it zooms in the middle of the image. Any way to zoom the image out, or to crop off the bottom of the image, as faces are often at the top?

jezs00, 2023-08-24:
You can play with some of the image generation settings, including the crop
to maybe see if that helps. If you're using dalle you can use infilling to
zoom out a bit - Those options are available using 'infill' and
'infill_percent' variables, have a play around with these if you want.

This functionality is currently not available on any of the other
providers.

Rocky56gh9, 2024-01-12:
> Thanks mate glad it is useful to you.
>
> 1. The bash script is hard to debug so this might be hard to fix. **If others have the same issue please post here so I can collect more information on it.** I [...]

I had the same issue with the folders not being created.

pabz007, 2024-01-31:
I'm having the same issue, any updates on how to go about it ? 
cp: cannot create regular file '/home/pi4/pycasso/prompts//artists.txt': No such file or directory
cp: cannot create regular file '/home/pi4/pycasso/prompts//subjects.txt': No such file or directory
cp: cannot create regular file '/home/pi4/pycasso/prompts//prompts.txt': No such file or directory
error: externally-managed-environment

Rocky56gh9, 2024-01-31:
> I'm having the same issue, any updates on how to go about it ?
>
> cp: cannot create regular file '/home/pi4/pycasso/prompts//artists.txt': No such file or directory
>
> cp: cannot create regular file [...]

Yes. The double slash meant it did not create the files. Had to create manually after creating the folders. Also had a permissions issue on the Config file. Up and running after that.

pabz007, 2024-01-31:
> Yes. The double slash meant it did not create the files. Had to create manually after creating the folders. Also had a permissions issue on the Config file. Up and running after that.

So you just created a "prompt" folder on pycasso? Do I need the Repository for Artist,  subjects, and prompts? I thought that was the OAI database when you add the API key.

Thanks for the help though!

Rocky56gh9, 2024-01-31:
> So you just created a "prompt" folder on pycasso? Do I need the Repository for Artist,  subjects, and prompts? I thought that was the OAI database when you add the API key.
>
>
>
> Thanks for the help [...]

I looked at the structure in the Examples folder. 

At the pycasso root folder, I created the images folder, then the generated and external folders. Then again at the root folder, I created the prompts folder and added the .txt files for artist, prompts, and subjects.

jezs00, 2024-01-31:
It's interesting that the bash file is not creating the files properly.
It's a bit harder to troubleshoot and test the bash file than the python
scripts, perhaps there's something in the operating environment that is
causing the path to be concatenated incorrectly.

jezs00, 2024-02-03:
Okay got it, as far as I can tell it's that the 
`PROMPTS_DIR="${LOCAL_DIR}/prompts/"
IMAGES_DIR="${LOCAL_DIR}/images/"`

should have been

`PROMPTS_DIR="${LOCAL_DIR}/prompts"
IMAGES_DIR="${LOCAL_DIR}/images"`

I guess it didn't pick up in testing as pycasso itself should also create the structure itself using backup files. In the case of this issue, it's likely an error that's a false flag distracting from file permissions, but an error nonetheless that should be fixed.

I still need to test this after upgrading my packages but it'd be cool if someone else could also test if the bash script is working on a new install. Otherwise if it's working for me I'll close the issue after a month of inactivity

jezs00, 2024-02-03:
@hplato FYI while working to update the openai API update I possible fixed a significant bug with the infill value not being passed properly to the infill function that needed it. Infill should work better now on the latest release if you're using dalle, and 

Please note openai has updated their api you should update your python packages or at least your openai package before updating pycasso.

pabz007, 2024-02-06:
> I looked at the structure in the Examples folder.
>
> At the pycasso root folder, I created the images folder, then the generated and external folders. Then again at the root folder, I created the [...]

It worked!, just waiting on the e-paper to come in tomorrow to give it a go. 

I troubleshoot the issue with the "externally-managed-environment". apparently is due to Ubuntu 23.04 or Fedora version 38. I was having trouble even running on venv which was odd. 

`cd /usr/lib/python3.11
sudo rm EXTERNALLY-MANAGED`

sorry, still learning! thanks all for the support! I'll be back if I run into more issues!

closed 2025-05-21 by jezs00