jezs00/pycasso #58: New: Text read from file rather than image prompt

djxie2019, 2023-08-07:
Would be great if there was an option to pull the text displayed from a file instead? Use case is using GPT to write descriptions.

jezs00, 2023-08-07:
Hey mate,

Can you be a bit more specific on the use case? The text displayed is
intrinsically linked to the image requested, so as far as I can picture it,
having a separate text file to draw text would be complicated to link up to
the request.

The only way I can see this working is either using GPT to write the
prompts pre-request, or implementing a GPT API in the application itself.

djxie2019, 2023-08-07:
Sorry I could have been clearer.

Upon boot I have a script that pulls a subjects.txt file from a Home Assistant instance prior to running Pycasso, which dynamically generates the prompt based on the weather, time of day and season. For example, full moon shown in the image if it's a full moon, or browning/bare trees in fall/autumn, raining if raining.

I want to also grab a dynamically generated text descriptor from Home Assistant to display underneath the image, that describes current conditions in more granular detail, ie. the temperature, forecast or any noteworthy happenings in the smart home that day. Home Assistant already has ChatGPT built in to generate these.

Would also be good to have more control of the text, ie. wrapping. Love what you've created btw.

jezs00, 2023-08-09:
Great idea. Yes, that's definitely possible. It will require config
updates, new file parser and appropriate insertion into the main program.
Should unit test it too. I'll try and update it at some point.

jezs00, 2023-08-27:
Should have the option to set override_text and override_path now in release 1.2.8.

commit 98943a7f (2023-08-27): Updated version. Closes #58.
closed 2023-08-27 by jezs00