jezs00/pycasso #59: Override text wrapping

djxie2019, 2023-09-02:
For the new text override function you've added, how can I add a line break in the text? In current state the string of text centres and if long enough, the start and end of the string are cut-off. If no line break character can create a new line, are you able to add a text wrap option?

Happy to send a donation for your time. Mentioned in other comments - using the frame as an AI-driven front end for a Home Assistant smart home.

Also what font format files are supported?

jezs00, 2023-09-03:
Unfortunately although possible, this would require a bit of work to implement in the code smoothly, and it kind of goes past the scope of the original project, probably only relevant for this very specific use case. I'm not sure when I'd have time/lust to work on this one in particular. I think your best bet is to do a pull request or a fork to work on a solution that works for what you're going for.

You would need to firstly modify the way add_text_to_image() works, but also the calculation of box limits in ImageFunctions.max_area() and possibly some cascading dependencies from that. Something along the lines of logging how many lines there would be, calculating box size including lines*fontsize, then repeating the draw.text as many times as appropriate with your lines.

Here are some resources that might help.
https://stackoverflow.com/questions/69371619/breaking-up-long-text-using-pillow
https://stackoverflow.com/questions/8257147/wrap-text-in-pil
https://www.geeksforgeeks.org/python-pil-imagedraw-draw-text/

pycasso uses Pillow to load the fonts, so you can find more details on what fonts are supported here: https://pillow.readthedocs.io/en/stable/reference/ImageFont.html

jezs00, 2025-05-17:
Well I mean it's not pixel art. But it does have multiple lines.

[image]

closed 2025-05-17 by jezs00