jezs00/pycasso #62: Error Building wheel for Pillow (PEP 517)

soldatino13, 2023-10-03:
me again...

Unfortunately I always get the same error when installing.
It doesn't matter whether I install on a Pi3 or Pi4.

[EPD Inky Impression 7 Color](https://shop.pimoroni.com/products/inky-impression)

What am I doing wrong?

`  Stored in directory: /tmp/pip-ephem-wheel-cache-tosihyjo/wheels/76/0e/97/ea5589bea962b7de0ae19c56590703f184b14973a355544b23
  Building wheel for waveshare-epd (setup.py) ... done
  Created wheel for waveshare-epd: filename=waveshare_epd-0.0.0-py3-none-any.whl size=134698 sha256=981828326550da4d5e9e1cf388505b1981d88194542ff3dfcb12e3f3f7a21ac3
  Stored in directory: /tmp/pip-ephem-wheel-cache-tosihyjo/wheels/3a/92/44/bc66898fdbd1f8eccb746a22638465e75f06c300d1771bec15
  Building wheel for Pillow (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /tmp/tmp94omurj2_in_process.py build_wheel /tmp/tmpdgyiuvf6
       cwd: /tmp/pip-install-kkcbvgeh/pillow_751f12e5f64f4cd7807f82219021391d
  Complete output (198 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-cpython-39
  creating build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageOps.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/DdsImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/TgaImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/GribStubImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/_binary.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/CurImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/TiffImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/EpsImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageQt.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageDraw.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageMorph.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageFilter.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageChops.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/features.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageEnhance.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/__init__.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageWin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/FitsImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/SgiImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/PixarImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/BdfFontFile.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageColor.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageMode.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ContainerIO.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/FtexImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/MpoImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/PaletteFile.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/BmpImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/PngImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/GifImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/PsdImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/Image.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageDraw2.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/WalImageFile.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/_deprecate.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/JpegImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/ImageTransform.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/PcdImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/PcfFontFile.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/FliImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/WmfImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/McIdasImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/BlpImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/Hdf5StubImagePlugin.py -> build/lib.linux-aarch64-cpython-39/PIL
  copying src/PIL/PdfParser.py -> build/lib.linux-aarch64-cpython-39/PIL
[... 127 lines cut ...]
    File "/tmp/pip-build-env-69ep4qej/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 434, in build_wheel
      return self._build_with_temp_dir(
    File "/tmp/pip-build-env-69ep4qej/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 419, in _build_with_temp_dir
      self.run_setup()
    File "/tmp/pip-install-kkcbvgeh/pillow_751f12e5f64f4cd7807f82219021391d/_custom_build/backend.py", line 47, in run_setup
      return super().run_setup(setup_script)
    File "/tmp/pip-build-env-69ep4qej/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 341, in run_setup
      exec(code, locals())
    File "<string>", line 1005, in <module>
  __main__.RequiredDependencyException:

  The headers or library files could not be found for jpeg,
  a required dependency when compiling Pillow from source.

  Please see the install instructions at:
     https://pillow.readthedocs.io/en/latest/installation.html

  ----------------------------------------
  ERROR: Failed building wheel for Pillow
Successfully built piblo omni-epd IT8951 openai stability-sdk waveshare-epd
Failed to build Pillow
ERROR: Could not build wheels for Pillow which use PEP 517 and cannot be installed directly
ERROR: Invalid requirement: '@'
ERROR: Invalid requirement: '@'
`

jezs00, 2023-10-04:
Hey, Pillow is a requirement for the image manipulation

Googling the error gets me https://stackoverflow.com/questions/61365790/error-could-not-build-wheels-for-scipy-which-use-pep-517-and-cannot-be-installe

So possibly try to run `pip3 install --upgrade pip` and then try again? If not, try using pip to uninstall `PIL` then try again, as I think they occupy the same namespace.

soldatino13, 2023-10-09:
@jezs00 : Great. Thank you very much. I was now able to install everything and it works perfectly.
Great Job!

closed 2023-10-09 by soldatino13