jezs00/pycasso #63: Generated images not working, test mode not working; External images working

Rocky56gh9, 2024-01-11:
1. Cannot get the automation to work to generate and display an image from DALLE.
2. Even with failover = True to display a test image in case of failure, test image never runs.
3. Putting an image in External folder can be displayed by running sudo systemctl restart pycasso.
4. Tried various configurations and cannot get automation to run.

Log and Config attached.
[config.txt](https://github.com/jezs00/pycasso/files/13909038/config.txt)
[pycasso.log](https://github.com/jezs00/pycasso/files/13909040/pycasso.log)

jezs00, 2024-01-12:
Check your install directory and path, could also possibly be a permissions issue. At least the last time you ran it, it couldn't find the config file and ran defaults, which would not output to the screen, just to mock epd. You can also check if that file is being created successfully.

`2024-01-11 17:04:16 INFO     /home/mini3/pycasso/.config does not exist. Copying /home/mini3/pycasso/examples/.config-example to /home/mini3/pycasso/.config`

Rocky56gh9, 2024-01-12:
> Check your install directory and path, could also possibly be a permissions issue. At least the last time you ran it, it couldn't find the config file and ran defaults, which would not output to the [...]

Interesting. I did note that if I run "ls" from the root folder, the .config file is not even visible. I have to run sudo nano to open it otherwise it says I don't have permission. I'll see if I can figure out how to unlock the root.

jezs00, 2024-01-12:
Yeah if that's the case and the directory paths in the error log are correct, then could likely be permission problems. Do a recursive chown and chmod to ensure that the user that runs pycasso has read write and execute for the directory and its contents.

Rocky56gh9, 2024-01-12:
That was it! The .config file's permissions were assigned to root rather than with everything else. I had to manually create the images and prompts folder structures, but it's running now.

This is really nifty!

drwxr-xr-x 9 mini3 mini3  4096 Jan 11 20:05 .
drwxr-xr-x 3 mini3 mini3  4096 Jan 11 19:31 ..
-rw-r--r-- 1 root  root   2205 Jan 11 20:05 .config
-rw-r--r-- 1 mini3 mini3  9429 Jan 11 19:31 .config_1
-rw-r--r-- 1 mini3 mini3   146 Jan 11 20:05 .creds
drwxr-xr-x 4 mini3 mini3  4096 Jan 11 19:31 examples
drwxr-xr-x 8 mini3 mini3  4096 Jan 11 19:31 .git
drwxr-xr-x 3 mini3 mini3  4096 Jan 11 19:31 .github
-rw-r--r-- 1 mini3 mini3  2053 Jan 11 19:31 .gitignore
-rw-r--r-- 1 mini3 mini3  1068 Jan 11 19:31 LICENSE
-rw-r--r-- 1 mini3 mini3   314 Jan 11 19:31 pycasso.service.template
-rw-r--r-- 1 mini3 mini3    80 Jan 11 19:31 pyproject.toml
-rw-r--r-- 1 mini3 mini3    77 Jan 11 19:31 pytest.ini
-rw-r--r-- 1 mini3 mini3 28128 Jan 11 19:31 README.md
-rw-r--r-- 1 mini3 mini3   210 Jan 11 19:31 requirements.txt
drwxr-xr-x 4 mini3 mini3  4096 Jan 11 19:31 resources
drwxr-xr-x 2 mini3 mini3  4096 Jan 11 19:31 scripts
-rw-r--r-- 1 mini3 mini3   982 Jan 11 19:31 setup.cfg
-rw-r--r-- 1 mini3 mini3    37 Jan 11 19:31 setup.py
-rw-r--r-- 1 mini3 mini3 11732 Jan 11 19:31 setup.sh
drwxr-xr-x 3 mini3 mini3  4096 Jan 11 19:31 src
drwxr-xr-x 7 mini3 mini3  4096 Jan 11 19:31 tests
mini3@mini3:~/pycasso $ sudo chown mini3:mini3 .config
mini3@mini3:~/pycasso $ ls -la
total 128
drwxr-xr-x 9 mini3 mini3  4096 Jan 11 20:05 .
drwxr-xr-x 3 mini3 mini3  4096 Jan 11 19:31 ..
-rw-r--r-- 1 mini3 mini3  2205 Jan 11 20:05 .config
-rw-r--r-- 1 mini3 mini3  9429 Jan 11 19:31 .config_1
-rw-r--r-- 1 mini3 mini3   146 Jan 11 20:05 .creds
drwxr-xr-x 4 mini3 mini3  4096 Jan 11 19:31 examples
drwxr-xr-x 8 mini3 mini3  4096 Jan 11 19:31 .git
drwxr-xr-x 3 mini3 mini3  4096 Jan 11 19:31 .github
-rw-r--r-- 1 mini3 mini3  2053 Jan 11 19:31 .gitignore
-rw-r--r-- 1 mini3 mini3  1068 Jan 11 19:31 LICENSE
-rw-r--r-- 1 mini3 mini3   314 Jan 11 19:31 pycasso.service.template
-rw-r--r-- 1 mini3 mini3    80 Jan 11 19:31 pyproject.toml
-rw-r--r-- 1 mini3 mini3    77 Jan 11 19:31 pytest.ini
-rw-r--r-- 1 mini3 mini3 28128 Jan 11 19:31 README.md
-rw-r--r-- 1 mini3 mini3   210 Jan 11 19:31 requirements.txt
drwxr-xr-x 4 mini3 mini3  4096 Jan 11 19:31 resources
drwxr-xr-x 2 mini3 mini3  4096 Jan 11 19:31 scripts
-rw-r--r-- 1 mini3 mini3   982 Jan 11 19:31 setup.cfg
-rw-r--r-- 1 mini3 mini3    37 Jan 11 19:31 setup.py
-rw-r--r-- 1 mini3 mini3 11732 Jan 11 19:31 setup.sh
drwxr-xr-x 3 mini3 mini3  4096 Jan 11 19:31 src
drwxr-xr-x 7 mini3 mini3  4096 Jan 11 19:31 tests

closed 2024-01-12 by Rocky56gh9

jezs00, 2024-01-12:
Thanks mate, have fun