jezs00/pycasso #64: Dalle Image generation not working

elkorrupto, 2024-02-02:
Hey there!

I´ve got problems with generating images with dalle and the logfile isn´t telling me what´s happening. 
The OpenAI API Key is working, i tried another script.

When i restart (sudo systemctl restart pycasso) the pycasso service my logfile looks like this:

pycasso.log:
2024-02-02 15:26:17 INFO     Config loaded
2024-02-02 15:26:17 INFO     Starting program without using PiJuice
2024-02-02 15:26:17 INFO     pycasso has begun
2024-02-02 15:26:17 INFO     Requesting 'La Sagrada Familia by David Hockney, digital art, trending on artstation'
2024-02-02 15:26:17 INFO     Loading Dalle API
2024-02-02 15:26:17 INFO     Getting Image

but then nothing is happening.

When i have a look at the status with:
sudo systemctl status pycasso

Henning@raspberrypi:~ $ sudo systemctl status pycasso
● pycasso.service - A system to send AI generated art to an E-Paper display through a Raspberry PI unit
     Loaded: loaded (/etc/systemd/system/pycasso.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2024-02-02 15:26:18 CET; 1s ago
    Process: 756947 ExecStart=dbus-run-session -- bash /home/Henning/pycasso/scripts/run.sh (code=exited, status=1/FAILURE)
   Main PID: 756947 (code=exited, status=1/FAILURE)
        CPU: 2.749s

Feb 02 15:26:17 raspberrypi dbus-run-session[756958]:   File "/usr/local/lib/python3.9/dist-packages/piblo/pycasso.py", line 703, in get_ima>
Feb 02 15:26:17 raspberrypi dbus-run-session[756958]:     self.image_base = self.load_dalle_image(self.prompt, self.width, self.height,
Feb 02 15:26:17 raspberrypi dbus-run-session[756958]:   File "/usr/local/lib/python3.9/dist-packages/piblo/pycasso.py", line 443, in load_da>
Feb 02 15:26:17 raspberrypi dbus-run-session[756958]:     image_base = dalle_provider.get_image_from_string(prompt, height, width)
Feb 02 15:26:17 raspberrypi dbus-run-session[756958]:   File "/usr/local/lib/python3.9/dist-packages/piblo/provider.py", line 262, in get_im>
Feb 02 15:26:17 raspberrypi dbus-run-session[756958]:     except openai.error.APIConnectionError as e:
Feb 02 15:26:17 raspberrypi dbus-run-session[756958]: AttributeError: module 'openai' has no attribute 'error'
Feb 02 15:26:18 raspberrypi systemd[1]: pycasso.service: Main process exited, code=exited, status=1/FAILURE
Feb 02 15:26:18 raspberrypi systemd[1]: pycasso.service: Failed with result 'exit-code'.
Feb 02 15:26:18 raspberrypi systemd[1]: pycasso.service: Consumed 2.749s CPU time.

The source files used are in "/usr/local/lib/python3.9/dist-packages/piblo/" and not in "/home/Henning/pycasso/src/piblo" as i expected - is that correct?

After installing by setup-script i had to create the folders ".../pycasso/prompts" and ".../pycasso/images" by myself, because they where missing.

I already tried 👍 to reinstall the hole raspberrypi, the pycasso installation as user and as sudo, up and downgrading openai and langchain versions to several version numbers, deleted ".error." from provider.py but nothing works at all.

What am i doing wrong?
Has anyone an idea? 
I´m sorry for my bad english and that i´m an absolute noob in raspberry - this is my first time...

Thank you so much for your time and ideas!
Henning

jezs00, 2024-02-03:
Annoying. Looks like dalle have updated the way their api works in a recent version of their python package. I'll see what I can do.

jezs00, 2024-02-03:
Dalle have changed a bit in how to use their script with release 1.0 of their API script so it doesn't work even with removal of errors:
```
You tried to access openai.Image, but this is no longer supported in openai>=1.0.0 - see the README at https://github.com/openai/openai-python for the API.

You can run `openai migrate` to automatically upgrade your codebase to use the 1.0.0 interface. 

Alternatively, you can pin your installation to the old version, e.g. `pip install openai==0.28`

A detailed migration guide is available here: https://github.com/openai/openai-python/discussions/742
```

We can hopefully fix this up with that migration guide

jezs00, 2024-02-03:
Alright I've got this working locally with the latest version of openai. Try using `bash <(curl https://raw.githubusercontent.com/jezs00/pycasso/main/setup.sh)` to update pycasso and let us know how it goes.

elkorrupto, 2024-02-03:
Oh my god, it´s working. 
Thank you so much!
You are a genius :)
I was struggeling for weeks and you did it in one night...
Thanks a lot from Germany - now i can play with the image size and colors...

I love this project!
Cheers ,
Henning

jezs00, 2024-02-03:
Ah. Ich lerne Deutsch. Es ist ein Problem, das schwer zu lösen ist, wann man hat Openai API noch nicht benutzt. Es ist leichter wann man hat Erfahrung in dieser Code. Sie waren nur unglücklich mit dem erste von dieser Problem. Viel glück.

Ich werde dieses Problem schließen.

Closing problem as solved

closed 2024-02-03 by jezs00