jezs00/pycasso #66: multiple generation lines

commo11, 2024-03-18:
Is there a way to use multiple lines in prompts.txt or have different prompt.txt files and have that randomly chosen with | as well? Sometimes I would like to see "A (happy | sad)(dog | cat)" other times I would like to see a "(winter | summer) (cabin | landscape)".

jezs00, 2024-03-18:
Hi commo,

This should be working out of the box. The prompts generator should always
pick a random line. See the readme for more information on formatting.

commo11, 2024-03-19:
I think I have it figured out, it all had to do with spaces and or lack of them, before a typical line would look like this "A (|angry|happy|sad) (|dog|cat|bird) (|eating|swimming|playing)". Sometimes this would work and I would get an image of A happy cat swimming with the correct text on the image, other times I would just get a random image back that had nothing to do with the words and a save name of just A .pycasso.png going to my NAS with absolutely no text on the display. Now I added spaces between the words and the | more like "A (| angry| happy| sad) (| dog| cat| bird) (| eating| swimming| playing)" and it seems to be working every time.
   
    Another question came up though, is it possible to save the file with the current date in it so they don't get over written if it does pick the same prompts?

jezs00, 2024-03-21:
> Another question came up though, is it possible to save the file with the current date in it so they don't get over written if it does pick the same prompts?

Should be relatively painless. I will add a new config item and put the functionality in.

jezs00, 2024-03-21:
Confirmed working on pi. Update minimally, migrate config, make sure save_date is set to True. This will add a date string to the start of your file names. On by default on new version

closed 2024-03-21 by jezs00