jezs00/pycasso #67: Idea for prompt

EA914, 2024-04-11:
Hi! I am going to try to do this but I was wondering if I could have a microphone connected and be able to dictate a prompt to be transcribed and send to DALLE to display in the frame, instead of having randomly generated AI art. Thoughts on how this could be implemented?

jezs00, 2024-04-13:
Cool idea. It's definitely possible. A few notes:

System would have to always be powered and the idea behind pycasso is to run and complete as quickly as possible, without staying on.

The solution that makes sense to me is:

1. Create/use a separate program or script that listens for your microphone, and when it does hear something, it calls pycasso.
2. pycasso needs to be updated to take a command line argument (and optional python arg if called in a python script) that overrides the random prompt with one provided to it. This should not be hard, just need to add an argument to be parsed and then based on its value, interrupt prompt generation with the argument.

I can handle 2 under this issue and create the functionality to override the prompt as I think this is in scope of this project and is functionality that others could use. As for 1 the microphone service, you'd have to implement a different project/library/program that allows this (probably loads of them for pis) and just modify it so it calls pycasso with args. That's something you'd have to do yourself as my implementation has no mic so I'd have no way to test if I wanted to.

EA914, 2024-04-13:
Thank you for responding so quickly! I have literally never worked with Raspberry Pi (or hardware at all), so I think for starters I'm going to try to replicate your project. I already have program that allows me to dictate a prompt through my computer's microphone, transcribe it with OpenAI's "whisper" model, and send the prompt to DALL-E to generate an image. In my mind, the Raspberry Pi would literally just run this program. How difficult would that be and what would it entail? You can see and run the code in my Github repo.

I'd just want a button connected via GPIO to the Raspberry Pi that, when held, allows you to dictate a prompt (also a red LED light indicator when it's held). When you let go of the button, it transcribes and sends it to DALLE for an image.

Another idea is to connect a small thumb-keyboard that will allow me to type a manual prompt instead of using my voice.

Any advice you have for me is greatly appreciated!

jezs00, 2024-04-14:
If you're already successfully grabbing an image then a hacky way to put it together would be to just write some scripting to save the image to the external images folder and ensure that it's the only image there. Then just set pycasso to external mode and run it once you've finished your own script.

Play around with ext info or names in the image files to get the title/artist on the page working if you want that.

EA914, 2024-04-14:
I'm gonna try that once my Raspberry Pi 4 arrives. I imagine having a button that triggers the microphone is just a matter of connecting it and configuring the drivers?

EA914, 2024-04-22:
In case you wanted to see my code and test this out with your own, this is what I have so far. You are the first to see it since you inspired me, this is not on github yet and i have not shared it with anyone.

Credits to https://github.com/DevMiser/AI_Art_Frame for also inspiring me and helping me get started. A lot of this base code is his.

1. The raspberry pi has a "wake word" of "Art Frame" that, when called, listens for your image prompt
2. The image is generated via dalle, and rendered on the inky display
3. The first GPIO button A will render the same image but with a text overlay showing the prompt (this was inspiration from you, and a lot of your code helped me out with this piece)
4. The second GPIO button will re-generate an image with the same prompt (in case you didn't like the first one dalle spit out)
5. The third GPIO button will show a QR code that points to the image URL so you can download the image on your phone

Please let me know your thoughts and how I can improve this!

```python
import datetime
import io
import openai
import os
import pvcobra
import pvleopard
import pvporcupine
import pyaudio
import random
import socket
import struct
import schedule
import sys
import threading
import time
import traceback
import urllib.request
import qrcode

from colorama import Fore, Style
from gpiozero import Button
from inky.auto import auto
from pathlib import Path
from openai import OpenAI
from PIL import Image,ImageDraw,ImageFont,ImageOps,ImageEnhance
from pvleopard import *
from pvrecorder import PvRecorder
from threading import Thread, Event
from time import sleep

import RPi.GPIO as GPIO
FONT_PATH = "Arial.ttf"
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
led1_pin=4
GPIO.setup(led1_pin, GPIO.OUT)
GPIO.output(led1_pin, GPIO.LOW)

audio_stream = None
cobra = None
pa = None
porcupine = None
recorder = None
wav_file = None

display = auto()

openai.api_key = "OPENAIAPIKEYHERE"
pv_access_key= "PICOVOICEAPIKEY HERE"

client = OpenAI(api_key=openai.api_key)

global img, img_resized, prompt_full, display_text_overlay, last_image_url	# Declare globals for image and prompt
prompt_full = ""  # Initialize the prompt variable
last_image_url = None

Clear_list = ["Clear",
[... 411 lines cut ...]

			event.set()
			GPIO.output(led1_pin, GPIO.LOW)
			sleep(2)
			t_refresh = threading.Thread(target=refresh_schedule, args=(event2,))
			t_refresh.start()

		else:
			print("Clearing the display...")
			clean_screen()
			event.set()
			event2.set()
			GPIO.output(led1_pin, GPIO.LOW)
			print("\nDone")

except Exception as e:
	print("Error:", e)

except ConnectionResetError:
	print ("Reset Error")
	current_time()

except KeyboardInterrupt:	 
	exit()
```

jezs00, 2024-04-25:
I think your code looks good (take this with a grain of salt because I'm no professional). Thanks for sharing it with me.

My recommendation is to do what I did and throw it up on github, maybe even do a little write up for some internet points and pop it up in the raspberry pi subreddit. If you're going to throw it up on github the only advice I'd give is abstract out your keys into another file so you can test easily and avoid posting your actual keys from a mistake when committing.

I will leave this issue open as it's overall a good idea and code wise there's quite a few good ideas that could be thrown into pycasso - however would take a bit of work and not having the tech it's harder for me to test.