jezs00/pycasso #68: Image scale issue

jackd248, 2024-08-21:
Hej there,

Thank you for the exciting project. 
I use a Raspberry Pi 3 with a 5.65 inch e-ink display. Unfortunately, the display is not completely filled with the generated image, two black borders remain. The images are generated in 512x512, the display resolution is 600x448 pixels. Is it possible for the image to be scaled completely to the edge? 

[image]

jezs00, 2024-08-21:
If you're using Dalle, recommend using infill = true for the edges. Costs a little more but looks better. Defined here in readme:
```
Generation

Settings related to generation of images with AI image providers

    image_rotate: Rotation of the image PRIOR to sending to providers. This way you can get an image that fits well in portrait or landscape as per your preference. (Integer)
    infill: A boolean flag that instructs pycasso to request an image to be infilled again if original image does not fill out the whole frame. (Boolean)
    infill_percent: If infill is set to true, this will make the original image request smaller by this percentage, and then infill the rest of the image to fit the frame. (Integer)

```

Let us know if that works.

jackd248, 2024-08-22:
You're right, I was looking for an upscale option within the display settings. 

Sorry for the misunderstanding.

closed 2024-08-22 by jackd248

jezs00, 2024-08-22:
No worries have fun