jezs00/pycasso #69: Stability AI API Change

Rocky56gh9, 2024-10-17:
The Stability AI API path has changed as of 10/11/24. The prior URL is no longer supported. I tried manually changing in all places that I found it to the new URL but there seems to be fundamental differences in how it works, so I'm not sure how to re-engineer independently. I am unable to get DALL-E to generate with an API key. In all cases, I'm just getting the test image now.

jezs00, 2024-10-18:
TLDR update `stable_host = "https://api.stability.ai/v1/generation/stable-diffusion-v1-6/text-to-image"` in .config and it should work again.

The API path or how it functions does not appear to have changed but the model that was in the path has been deactivated. We should start using stable-diffusion-v1-6 instead. I have updated this and tested as working. From https://platform.stability.ai/docs/api-reference#tag/SDXL-1.0-and-SD1.6/operation/textToImage:
`SD1.6 is a flexible-resolution base model allowing you to generate non-standard aspect ratios. The model is optimized for a resolution of 512 x 512 pixels`

If you try to use one of the larger/XL models it won't work as the image size is too big for our resize operations. The smaller models work better as e-paper displays tend to be smaller than the 1024*1024 that XL models like, so IMO refactoring the stable diffusion code to allow for XL models wouldn't be a priority (saying that, happy for someone to merge to implement if they like, it shouldn't be too hard).

As for dalle, I am not sure, can you please provide logs and your config? I have confirmed dalle is still working on my implementation.

Please let me know if the above works.

Rocky56gh9, 2024-10-18:
Thank you! The "stable_host" change works.

Re: Dall-E, the logs consistently show it just stopping at "Getting Image." It never progresses beyond that. I'm not too concerned as I prefer the Stability.AI output anyway, so I'm happy with that working again.

2024-10-17 10:04:43 INFO     Config loaded
2024-10-17 10:04:43 INFO     Starting program without using PiJuice
2024-10-17 10:04:43 INFO     pycasso has begun
2024-10-17 10:04:43 INFO     Requesting 'A Reverent individual Photographing through Colonial Inn, feeling the weight of Memories, while the Dawn cast>
2024-10-17 10:04:43 INFO     Loading Dalle API
2024-10-17 10:04:43 INFO     Getting Image

jezs00, 2024-10-19:
Will need to keep an eye on that dalle issue, it looks like possible something to do with string parsing. I'll close this issue off for now.

closed 2024-10-19 by jezs00