jezs00/pycasso #70: Stability engine depreciation (stable-diffusion-xl-beta-v2-2-2)

hplato, 2024-11-18:
Hello, I just updated to the latest hoping this might be fixed, but it looks like pycasso uses an older stability.ai engine. I now get this error:

2024-11-17 18:07:37 INFO     Using https://api.stability.ai/v1/generation/stable-diffusion-xl-beta-v2-2-2/text-to-image as stability host
2024-11-17 18:07:37 INFO     Getting Image
2024-11-17 18:07:37 DEBUG    Starting new HTTPS connection (1): api.stability.ai:443
2024-11-17 18:07:38 DEBUG    https://api.stability.ai:443 "POST /v1/generation/stable-diffusion-xl-beta-v2-2-2/text-to-image HTTP/1.1" 404 None
2024-11-17 18:07:38 ERROR    Non-200 response: {"id":"xxxxx","message":"The specified engine (ID stable-diffusion-xl-beta-v2-2-2) was not found.","name":"not_found"}

It looks like that engine was depreciated on Oct 11th:

https://platform.stability.ai/docs/release-notes

I'm not sure if stable-diffusion-xl-1024-v1-0 or stable-diffusion-v1-6 would be the better engine going forward.

jezs00, 2024-11-18:
Good Afternoon,

I believe this has been noted and fixed in #69 . Check it out for more information on the issue and fix.

TLDR update `stable_host = "https://api.stability.ai/v1/generation/stable-diffusion-v1-6/text-to-image"` in .config and it should work again.

Please confirm if this works.

hplato, 2024-11-18:
Yes, thank you. That seemed to work. my .config file is from an old version and didn't have the stable_host entry. I'm guessing the .config file wasn't touched during the upgrade to the latest.

closed 2024-11-18 by hplato

jezs00, 2024-11-26:
I've actually updated the constant in code now, so this should help for anyone on an old .config file. No guarantees it won't be updated again later, but this should help out for now.