jezs00/pycasso #71: New idea for prompts?

BigDave365, 2025-04-05:
I have had my screen up for a year and love it. I never fails to have something new and unexpected displayed.

Have you given an thought of having the Pi reach out to a random news feed, parse a random headline, and use that as the prompt?

Given the crazy news lately, this could lead to some interesting interpretations of current events.

I am not much of a programmer, so am of little help...sorry.

Anyhow, great project and thanks so much for making in available!

jezs00, 2025-04-06:
Can probably hack this together quite easily. Maybe slightly harder to do well. I haven't touched the project for a while so might be fun to try again. 
* new prompt mode - read from RSS feed - let's use [feedparser](https://pypi.org/project/feedparser/)
* take top item and use as subject
* set up fallbacks for failure on RSS read

BigDave365, 2025-04-06:
Perhaps have it add a prefix or suffix phrase from one of the lists. So if feedparser returned "Cow destroys shopping mall"...the prefix would be "An oil painting of" or "A 3D render of" etc....

I can only imagine the things that might come out.

The only hitch I see is many AI image generation platforms have guard rails for politically sensitive images and it might return null if the headline is such. IDK. Worth a shot if its easy...

lennartvandeguchte, 2025-04-21:
Hi both, 

First of all, thanks @jezs00 for building this great project. I forked your project a while ago and setup my raspberry pi plus e-ink screen with the idea to add some additional image generation modes like the ones you discuss here.  However, since then I left it on the shelf for while but yesterday I took the time to create a script to experiment with different image generation modes. 

My current workflow that (sometimes) gives some nice results:

1. Retrieve a random philosophical quote from https://zenquotes.io/api/random (however, this could be easily replaced by a news headline)
2. Enhance this quote by throwing it in a LLM and prompt the model such that the output should be a prompt that suits an image generation model.
3. Use the image generation prompt from step 2 to feed to an image generation model. 
4. Have again a LLM generate an explanation about how the image is representing the philosophical quote using creative writing (would actually be nice to use a VLM for this, then we can start using an AI as a (art) judge :). In the end I was thinking to have some kind of AI agent that uses feedback (maybe human feedback as well) to improve upon itself over time by reviewing its own generated images. ) 

Do you guys have other ideas besides the news headlines I could try out? 

I will try to implement the above workflow in the codebase. First into my forked version and then I could try to make a pull request if you're interested?

lennartvandeguchte, 2025-04-21:
Here an example of the above workflow to give you an idea :) 

1. Quote: "Wit is educated insolence." - Aristotle
2. LLM Generated Prompt: Amidst a surreal landscape of floating spheres, depict a figure exuding confidence and playfulness, their features a blend of ancient wisdom and mischievous charm. Render this in a style reminiscent of surrealist digital art, with vibrant colors and intricate details that hint at hidden depths. Capture a mood of enigmatic brilliance and audacious intelligence.
3. Generated image:
[image]
4. Explanation: In this mesmerizing digital artwork, the essence of Aristotle's profound quote, "Wit is educated insolence," materializes with bewitching clarity. The surreal landscape of floating spheres embodies the boundless realm of intellectual exploration, where knowledge defies gravity and conventional constraints. At the heart of this enigmatic cosmos stands a figure exuding a potent blend of ancient wisdom and mischievous charm, their very presence a testament to audacious intelligence.

The figure, rendered with intricate details and vibrant colors, exudes confidence and playfulness in equal measure. Their features, a reflection of both profound insight and irreverent wit, speak to the duality of educated insolence. The hidden depths hinted at in the artwork mirror the layers of intellect and cunning that lie beneath the surface of wit. Each sphere in the surreal landscape represents a facet of knowledge, floating in harmony yet distinct, hinting at the multifaceted nature of intelligence.

The mood captured in this artwork is one of enigmatic brilliance, inviting viewers to ponder the fine line between bold intelligence and impudent wit. The surrealist style employed here elevates the philosophical essence of the quote, offering a visual symphony of thought-provoking symbolism. As we gaze upon this digital masterpiece

jezs00, 2025-04-21:
Looks awesome. At first I was just going to set this up just for the hell of it, but now I actually want this as an option for my frame too because it looks sweet. I have been planning on working on this but been a little busy, might have some good time to implement some stuff this weekend. 

Without checking out your fork, I think this should be implemented in a way that is easily extensible for others who want to implement any kind of prompt generation in code. With this in mind, happy for anyone to pull and throw some code together, but might be best to wait for some structure to be set up (or set it up yourself) so we can do it in a standard way. In my opinion anyway. 

Proposal
* Create new superclass "Block" (better name tbc maybe) 
* Classes that extend Block must also extend a function generate() that returns text
* Sketchy plan is to allow stuff like "<rss> <weather> <llmsummary>" work from text fields. This will need some thinking, best method I can see right now is pull text, switch the string and use that to call appropriate function. Also consider passing parameters somehow. 
* Noting any time we go to an external provider for functionality we have a big risk of failure so we must have good error handling and appropriate fall backs
* Taking the above and then sending it to an llm for further processing may be a little difficult but passing parameters may be the way around this. Want to make it as modular as possible for those who won't be writing their own scripts. 

I'm just typing this on my phone pondering things so implementation might be different, but I think first thing I'll do is create Block or whatever to hopefully allow whoever to pull, extend and create their own scripting very easily. Will set up unit tests and implement RSS and weather for examples on how it works.

BigDave365, 2025-04-21:
Here is an Idea:

How about a selection of texts...one could have classic books, movie scripts, transcripts. Randomly pick a sentence in a random book and use that as a prompt. 

Books like "The Hobbit" would yield good results.

lennartvandeguchte, 2025-04-23:
Sounds like a good plan to me. Looks the below code like what you had in mind?

`class PromptBlock(abc.ABC):
    """
    Abstract base class for all prompt generation blocks.

    Subclasses must implement the `generate` method, which should return
    a string representing the content block.

    Error handling for external dependencies (e.g., APIs for weather, RSS) 
    should be implemented within the `generate` method of the respective subclasses.
    Consider appropriate fallbacks or raising specific exceptions.
    """

    @abc.abstractmethod
    def generate(self) -> str:
        """
        Generates the text content for this block.

        Returns:
            str: The generated text content.

        Raises:
            NotImplementedError: If the subclass does not implement this method.
            Exception: Subclasses should handle and potentially raise exceptions
                       related to their specific data fetching or processing logic.
        """
        raise NotImplementedError("Subclasses must implement the generate() method.")

    def __str__(self) -> str:
        """Provides a default string representation, useful for debugging."""
        return f"{self.__class__.__name__}()"

    def __repr__(self) -> str:
        """Provides a detailed string representation."""
        return f"{self.__class__.__name__}()"

class ZenBlock(PromptBlock):
    """
    A prompt block that generates text containing a random Zen quote.
    Fetches data from https://zenquotes.io/
    """
    API_URL = "https://zenquotes.io/api/random"
    TIMEOUT = 10 # seconds

    def __init__(self):
        # No specific initialization needed for this block
        pass

    def generate(self) -> str:
        """
        Fetches a random Zen quote and returns it.

        Returns:
            str: The Zen quote string, or an error message/fallback.
        """
        try:
            logging.info(f"Fetching random zen quote from {self.API_URL}...")
            response = requests.get(self.API_URL, timeout=self.TIMEOUT)
            response.raise_for_status() # Raise HTTPError for bad responses (4xx or 5xx)
            data = response.json()

            if data and isinstance(data, list) and len(data) > 0:
                quote_data = data[0]
                quote = quote_data.get('q')
                author = quote_data.get('a')
                if quote and author:
                    full_quote = f'\"{quote}\" - {author}'
                    logging.info(f"Successfully fetched quote: {full_quote}")
                    return full_quote
                else:
                    logging.error("API response missing quote or author.")
                    return "[Error: Could not parse quote data]"
            else:
                logging.error(f"Unexpected API response format from ZenQuotes: {data}")
                return "[Error: Unexpected quote API response]"

        except requests.exceptions.Timeout:
            logging.error(f"Timeout occurred while fetching Zen quote from {self.API_URL}")
            return None
        except requests.exceptions.RequestException as e:
            logging.error(f"Error fetching zen quote: {e}")
            return None
        except Exception as e:
            # Catch any other unexpected errors during processing
            logging.exception(f"An unexpected error occurred processing the ZenQuotes API response: {e}")
            return None
 `

The only thing I didn't understand is what you mean with: 
- "Sketchy plan is to allow stuff like "<rss> <weather> <llmsummary>" work from text fields. This will need some thinking, best method I can see right now is pull text, switch the string and use that to call appropriate function. Also consider passing parameters somehow." 

I suggest to move all prompt generation stuff (including the existing) to a file named `promp_generation.py`. I will try to make this work now for my 'quote' workflow and let you know when it's working so you can have a look yourself.

lennartvandeguchte, 2025-04-23:
The code can found on this branch https://github.com/lennartvandeguchte/pycasso/tree/feature/add_new_generation_mode

When comparing it to your main branch it also includes several changes I had to implement to make the code work for me. And it also includes some extra test scripts now. However, the main code for the workflow we discussed can be found in prompt_generation.py and pycasso.py. 

The simple workflow for fetching a Quote and using this as a prompt now works on my raspberry pi :) If you like I can start making a pull request and we can we pick it up from there.

jezs00, 2025-04-23:
Awesome throw up a pull request and we can work on integrating it - just want to make sure it's easy for others to use similar workflows. I can probably work on it this weekend. Don't worry about the sketchy plan, it should be a pretty minor thing to set up.

jezs00, 2025-04-24:
Pull request and feature branch up.
Once merged - TODO:
* Dynamic prompt integration
* Also integrate RSS feed
* Also integrate weather

BigDave365, 2025-04-25:
Wow, great to see you guy breathing some life into this. When you are done, would you please write a short "how to" to update and use the new features? I am not that talented with a RPi, but I can follow directions...;)

jezs00, 2025-04-30:
Progress - working somewhat well. Excuse the font size.

This was generated in SD from the text
`{<quote>} in the style of <file:prompts/artists.txt>`

[image]

Basically, we can use <> brackets to utilise blocks in our prompts, and we can also set the blocks up to take input as parameters using <block:argument:argument> etc.

Because this might get a bit too wordy to display as text on the box you can also use {} brackets to specify what you want to actually show on the screen. So in the example above we are only using the quote as the title text. To be updated in readme when it's all working well.

The whole thing needs some massaging, what is pending for this issue is:

- LLM block
- RSS block

Future work:
- Joke block
- Weather block
- Status icons based on the above results

@lennartvandeguchte I've invited you as a collaborator, have a play around on this branch to try to get LLM block working. Main files to be worked on is prompt_block.py and prep_prompt_text() in pycasso.py

Have fun!

jezs00, 2025-04-30:
Something I should also mention before you dip in, you can nest blocks, which will allow you to pass results from one block to another. Works inside - out. The intended functionality is to make it so people can do complicated constructions just using the prompt input. 

EG
<llm:write a prompt for image generation that evokes the feeling of the quote <quote>>

The quote will fill first, passing the full phrase to the llm, which will in turn be replaced with the generated prompt.

lennartvandeguchte, 2025-04-30:
Aha I now understand what you mean with the nested blocks, looks good! That indeed adds a lot of flexibility to the configuration of the prompt.

jezs00, 2025-05-03:
Hey @lennartvandeguchte did you end up working on anything to do with llm blocks that needs to be pushed? Otherwise I was considering working on that today to merge the branch and track other proposed blocks in separate issues.

lennartvandeguchte, 2025-05-04:
Hi @jezs00 , I have tested your code and it works for me. Furthermore I started with the LLM block but nothing serious yet. Unfortunately, I only have time this Wednesday to work on it so please go ahead if you like to implement it yourself.

jezs00, 2025-05-04:
Ah I got distracted this weekend anyway and did nothing so jump on if you still want to, otherwise I'll pick it up next weekend.

jezs00, 2025-05-08:
> I've added a LLM block with openai as the LLM provider. It's still work in
> progress since I got a bit confused about where and how config variables
> are defined. I tried to add an extra API key in a similar way as is done
> for the other providers but it doesn't work yet, therefore you will find a
> TODO statement in the code in which I defined this api key manually as a
> environment variable for the time being.
>
> Hopefully I find some time this weekend to finish it off but feel free to
> continue and improve. Some things to consider/implement:
>
>    - Do you want separate API keys for dall-e and a LLM, or have a single
>    one for OpenAI?
>    - Move api call from LLMBlock to LLM provider class
>    - review config variables
>    - tests

Awesome. I think we should use the same api key if it works, probably keeps
things simpler. Config setup is a little arduous just because of the
fallbacks and tests so I can throw that together relatively quickly. If you
want to push to the branch I can see about massaging it together to merge
with some instructions.

lennartvandeguchte, 2025-05-08:
Ah classic, I forgot to push. I have now pushed the code and refactored it a bit such that it uses the dalle key (might want to change this to openai).

jezs00, 2025-05-11:
Looking good! Definitely a few todos to sort out but I'll get the RSS done today I think to close this one out and let people play with it after publishing as a new release

A few todos extending on your list:

- Now that the text generated can be unpredictable, we might like to better support text wrapping and dynamic resizing, track this under #59 
- More tests, I really like the responses.activate that @lennartvandeguchte used and I want to extend this to all providers #74
- Weather #75
- Dad Joke #76

[image]

jezs00, 2025-05-11:
Alright that was fun. I'm fixing up the readme now, then will push and release it so everyone can mess around with it.

[image]

jezs00, 2025-05-11:
See new readme for instructions on using the prompt block and rss.

lennartvandeguchte, 2025-05-11:
Awesome! 👏 👏 👏