jezs00/pycasso #82: Need a bit of help sorting out RSS feed format.

BigDave365, 2025-05-21:
I am having some issues trying to sort out the format for the RSS feed block.

My prompt 

<rss;https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml;title;3> in the style of Peter Max

Returns nothing...

<<rss;https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml;title;3> in the style of Peter Max>

Yields :

2025-05-21 09:33:40 WARNING  "house g.o.p. races to revamp major policy bill, grasping for votes to pass it in the style of peter max" not found, please check readme for valid blocks. Using blank 

Any ideas?

jezs00, 2025-05-21:
When you say it returns nothing for the first prompt, does it return just
"in the style of Peter Max" or a completely blank string?

Confirming that text input without blocks works? Can you run it and post
your log file?

The warning implies that the block itself is working and *should* be
working in the first prompt. Possibly also try testing out <quote> and
<joke> blocks as well to see if you have similar issues with those.

BigDave365, 2025-05-22:
By nothing, nothing was returned and a blank was sent to Dalle.

However, for whatever reason, it started working when I went back to look at this this afternoon. Everything once in a while, it will return a blank (last time it started from the Cron job it was blank). 

I don't think it is worth anybody's time to chase whatever the issue is down...

These enhancements are great fun really add great potential...

Now to figure out the the text formatting (The longer prompts get truncated by the edge of the screen)...

jezs00, 2025-05-23:
So a couple of things, one of which we can work on. First the question
`Now to figure out the the text formatting (The longer prompts get truncated by the edge of the screen).`
Take a look at the readme file, particularly the new text wrapping variables. Text wrapping and dynamic resizing was added in #59 recently. Play around and see what works for your screen. Make sure you also try using `subject_brackets` setting with {} to specify a subject and keep your text size down.

https://private-user-images.githubusercontent.com/86394236/444759979-180d6cb3-76fc-4d6c-a9be-dc510ad007c4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDc5ODY5OTYsIm5iZiI6MTc0Nzk4NjY5NiwicGF0aCI6Ii84NjM5NDIzNi80NDQ3NTk5NzktMTgwZDZjYjMtNzZmYy00ZDZjLWE5YmUtZGM1MTBhZDAwN2M0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTIzVDA3NTEzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJiMzRhZmNlNjQ2ZTAzMzg0NjU5ZmNjYTg0M2FjZTg3ODljYThmMjE2ZTczMTJlNmFjZDBmNTVjNzgyMzhkNWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.DpoRxXt39TdRmCnTBkkuWtm82j0FNi1hn8fXaTBycs8

jezs00, 2025-05-23:
Note that returning a blank string is expected functionality on a failed block fetch. So in general if you are getting blank strings, then you may have problems with the block provider(s). As we are going out to one or many services in the web, a lot can go wrong with that and on failure we currently just have the block return blank so that at least the rest of the prompt can be attempted. If you are getting routine failures, it might be anything from a problem with this program, the format of the rss, the arguments being provided, etc. You will have to keep an eye on the log.

However, this being said we can probably add a fallback mode for all block types - EG if a block fails to load, then try a different block or try a prompt with no blocks.

It will add a little complexity to the way we do prompts, but at its simplest it could just be a string that we pass to any block, so that it will return that string if the block fails. Otherwise we could have a 'fallback' file and if we don't manage to get a block working from our main file, we use that.

I'll keep this issue open until fallback blocks are implemented.